kraj/meta-openwrt #3: README.md does not reflect the right version line in the sand for this layer.

madscientist42, 2016-06-03:
ERROR: No recipes available for:
  /home/frank/git/oe-build/build/../meta-openwrt/recipes-tweaks/iptables/iptables_1.6%.bbappend

From a Jethro based checkout : 

find . -name iptables*bb
./meta/recipes-extended/iptables/iptables_1.4.21.bb

/git/oe-build/meta$ git branch
* jethro

Jethro's 2.0.1...  If you meant "Krogoth" (2.1) or later, that'd be fine and should be reflected in the docs.  I can very probably move to Krogoth with no real consequence in my project, but you probably ought to either adjust the tweak recipes or document the requirement a bit more precisely...

kraj, 2016-06-14:
Patches are different for iptables 1.6, if you remove that bbappend then it should work fine. However We do not have jethro branch, which implicitly means that it was never tested with jethro. We will have krogoth however.

madscientist42, 2016-06-14:
Works.  Perhaps a minor document update so people don't trip up on it.  One
would like an undistilled version of what you're working to.   ;-)

Right now I'm trying to see if the woes I have are due to GCC or not.  May
be some more gotchas to be logged on the way

closed 2016-06-14 by kraj