kraj/meta-openwrt #4: luci doesn't build

ambrice, 2017-01-05:
Using yocto morty, "./scripts/feeds install -a" fails with:

DEBUG: Executing shell function do_compile
NOTE: make -j 8 TOPDIR=/home/brice_a/projects/morty/1233build/tmp/work/armv7a-neon-poky-linux-gnueabi/luci/git-r0/git/openwrt
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... failed.
Checking 'g++'... ok.
Checking 'working-g++'... failed.
Checking 'ncurses'... failed.
Checking 'zlib'... failed.
Checking 'libssl'... failed.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'md5sum'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python'... ok.
Checking 'svn'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'openssl'... ok.
Checking 'ldconfig-stub'... ok.

Build dependency: Please reinstall the GNU C Compiler - it appears to be broken
Build dependency: Please reinstall the GNU C++ Compiler - it appears to be broken
Build dependency: Please install ncurses. (Missing libncurses.so or ncurses.h)
Build dependency: Please install a static zlib. (Missing libz.a or zlib.h)
Build dependency: Please install the openssl library (with development headers)

/home/brice_a/projects/morty/1233build/tmp/work/armv7a-neon-poky-linux-gnueabi/luci/git-r0/git/openwrt/include/prereq.mk:12: recipe for target 'prereq' failed
Prerequisite check failed. Use FORCE=1 to override.
/home/brice_a/projects/morty/1233build/tmp/work/armv7a-neon-poky-linux-gnueabi/luci/git-r0/git/openwrt/include/toplevel.mk:151: recipe for target 'staging_dir/host/.prereq-build' failed
make[1]: *** [staging_dir/host/.prereq-build] Error 1
/home/brice_a/projects/morty/1233build/tmp/work/armv7a-neon-poky-linux-gnueabi/luci/git-r0/git/openwrt/include/toplevel.mk:80: recipe for target 'prepare-tmpinfo' failed
make: *** [prepare-tmpinfo] Error 2
Cannot open './tmp/.packageinfo': No such file or directory
Can't open file './tmp/.targetinfo': No such file or directory
WARNING: No feed for package '–a' found, maybe it's already part of the standard packages?
Ignoring feed 'packages' - index missing
Ignoring feed 'luci' - index missing
Ignoring feed 'routing' - index missing
Ignoring feed 'telephony' - index missing
Ignoring feed 'management' - index missing
Ignoring feed 'targets' - index missing
DEBUG: Shell function do_compile finished

mattsm, 2017-01-05:
Probably useful to include branches you are using for all components as well as the actual bitbake commands

ambrice, 2017-01-05:
I am using the morty branches of meta-openembedded, poky, meta-nodejs, and meta-freescale.

I added luci to a custom image recipe.  It seems to succeed, bitbake doesn't report any errors but it creates an empty package, and looking at the do_compile.log above it seems like it's silently failing to compile.  "bitbake luci" does the same.

meta-openwrt does not have any branches, what branch was it developed with?

kraj, 2017-01-19:
I expect it to work with morty and jethro as well as master. If there are ABI breaking changes then we can branch. Please let me know

kraj, 2017-01-23:
@ambrice can you retry with recent fixes on master and update if this issue is fixed now ?

troth, 2017-01-26:
Just tried replacing luci2 with luci in the new openwrt-base-image.bb recipe and hit the same problem. Investigating it now...

troth, 2017-01-26:
Found the following in the recipe:

```
$ echo '–a' | od -c
0000000 342 200 223   a  \n
0000005

$ echo '-a' | od -c
0000000   -   a  \n
0000003
```

Looks like some unicode got pasted into the recipe.

```
@@ -22,8 +22,8 @@ EXTRA_OEMAKE += "TOPDIR=${S}/openwrt"
 
 do_compile () {
        cd ${S}/openwrt
-       ./scripts/feeds update –a
-        ./scripts/feeds install –a
+       ./scripts/feeds update -a
+       ./scripts/feeds install -a
 
        for d in modules/luci-base
        do
```

With the above change I get a little further in the installation of the feeds, but  still get the following errors when building openwrt-base-image:

```
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: openwrt-base-image-1.0-r0 do_rootfs: luci not found in the feeds (qemux86 i586 x86 noarch any all) in /home/troth/git/econe-yocto/build-morty-qemux86/tmp/deploy/rpm.
ERROR: openwrt-base-image-1.0-r0 do_rootfs: This is often caused by an empty package declared in a recipe's PACKAGES variable. (Empty packages are not constructed unless ALLOW_EMPTY_<pkg> = '1' is used.)
ERROR: openwrt-base-image-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/troth/git/econe-yocto/build-morty-qemux86/tmp/work/qemux86-poky-linux/openwrt-base-image/1.0-r0/temp/log.do_rootfs.8447
ERROR: Task (/home/troth/git/econe-yocto/build-morty-qemux86/../meta-openwrt/recipes-core/images/openwrt-base-image.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2704 tasks of which 2703 didn't need to be rerun and 1 failed.
```

Here's the log from do_compile of the luci package:

```
+ for d in modules/luci-base
+ cd /home/troth/git/econe-yocto/build-morty-qemux86/tmp/work/i586-poky-linux/luci/git-r0/git/modules/luci-base
+ oe_runmake
+ oe_runmake_call
+ bbnote make -j 8 TOPDIR=/home/troth/git/econe-yocto/build-morty-qemux86/tmp/work/i586-poky-linux/luci/git-r0/git/openwrt
+ '[' -p /home/troth/git/econe-yocto/build-morty-qemux86/tmp/work/i586-poky-linux/luci/git-r0/temp/fifo.30539 ']'
+ printf '%b\0' 'bbnote make -j 8 TOPDIR=/home/troth/git/econe-yocto/build-morty-qemux86/tmp/work/i586-poky-linux/luci/git-r0/git/openwrt'
+ make -j 8 TOPDIR=/home/troth/git/econe-yocto/build-morty-qemux86/tmp/work/i586-poky-linux/luci/git-r0/git/openwrt
+ ret=0
+ trap '' 0
+ exit 0
DEBUG: Shell function do_compile finished
```

Still investigating...

mattsm, 2017-01-26:
Is anything staged for packaging for luci? Check in the packages-split dirs in tmp/work.../luci dir

troth, 2017-01-26:
Nothing in packages-split:

```
$ find packages-split/
packages-split/
packages-split/luci-doc
packages-split/luci-dev
packages-split/luci-dbg
packages-split/luci-staticdev
packages-split/luci
packages-split/luci-locale
```

I think (guessing from looking at the code) the problem is trying to use the openwrt build system with oe_runmake. Luci is not in the openwrt/.config file, so the build does nothing and nothing ends up in the openwrt/staging_dir.

troth, 2017-01-26:
I've done a rewrite of the luci recipe that just manually copies the luci files into the dest dir (they're all just files that don't need compiling anyways) and avoids using the openwrt build system. Here's my current version of the luci_git.bb recipe:

```
# Copyright (C) 2015 Khem Raj <raj.khem@gmail.com>
# Released under the MIT license (see COPYING.MIT for the terms)

DESCRIPTION = "OpenWrt system message/RPC bus"
HOMEPAGE = "http://git.openwrt.org/?p=project/libubox.git;a=summary"
LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENSE;md5=2b42edef8fa55315f34f2370b4715ca9"
SECTION = "base"
DEPENDS = "json-c libubox lua5.1"
RDEPENDS_${PN} = "lua5.1"

SRCREV_luci = "5b79e62c0a99bab8dfb8dce8124d9fecc11da54b"
SRCREV_openwrt = "v15.05.1"

SRC_URI = "git://github.com/openwrt/luci.git;name=luci \
"

inherit pkgconfig openwrt

S = "${WORKDIR}/git"
EXTRA_OEMAKE += "TOPDIR=${S}/openwrt"

FILES_${PN} = "/www \
               /usr/share \
               /sbin \
               /etc/config \
"

do_configure[noexec] = "1"
do_compile[noexec] = "1"

do_install () {
	cp -a ${S}/modules/luci-base/htdocs ${D}
	mv ${D}/htdocs ${D}/www
	chown -R root:root ${D}/www

	install -Dm 0644 ${S}/modules/luci-base/root/www/index.html ${D}/www/index.html

	install -Dm 0644 ${S}/modules/luci-base/root/etc/config/ucitrack ${D}${sysconfdir}/config/ucitrack
	install -Dm 0644 ${S}/modules/luci-base/root/etc/config/luci ${D}${sysconfdir}/config/luci

	install -Dm 0755 ${S}/modules/luci-base/root/sbin/luci-reload ${D}/sbin/luci-reload
	install -Dm 0644 ${S}/modules/luci-base/root/usr/share/acl.d/luci-base.json ${D}/usr/share/acl.d/luci-base.json
}
```

I will submit this in an upcoming pull request with a bunch of other changes I have pending. I've been working on getting the openwrt-base-image to boot to a semi-functional openwrt chaos-calmer system. I'm still working on getting netifd to startup and setup eth0.

kraj, 2017-01-26:
This looks ok to me. Submit it as a patch.

troth, 2017-01-27:
I've pushed a branch to my fork with the changes, but I have not been able to test if it is functional yet (don't have networking up yet in my image). All I can say is that luci builds and the files get installed in the image. Will submit a pull request when I have been able to test that luci works in a browser connected to my image.

kraj, 2017-01-27:
sounds ok

ambrice, 2017-01-27:
Updated recipe builds for me, but also haven't gotten the prerequisites running.  What is the status of luci2, is it stable and full-featured?  I was trying to get luci working since that's what seems to be in default chaos calmer, but I haven't found a lot of recent status on luci2.

kraj, 2017-01-27:
I dont think we have working luci2 setup. Its not default in openwrt too IIRC

troth, 2017-01-27:
It doesn't look like luci2 has seen any dev activity in nearly two years. I'm assuming that it is dead and since luci has seen recent activity that it is still moving forward. I already changed the openwrt-base-image to use luci instead of luci2 and once I get luci tested and work, will likely submit a patch to remove luci2 completely from the meta-openwrt layer.

closed 2017-02-07 by ambrice