kraj/meta-openwrt #15: add list of known working boards to a README or BOARDS file

mattsm, 2017-02-28:
Pretty self explanatory... but good info for those looking to get started / test.

I have a vague desire to add support for a QCA ar71xx MIPs board...

troth, 2017-03-04:
Khem and I talked about specifying a reference board for development use and he suggested we use a RaspberryPi 3:

- Cheap and near universal availability.
- Has ethernet port on board.
- Has Wifi on board (although need to determine if it supports AP mode).

As things stand currently, I think that Aaron Brice is the only one that has gotten wifi working so far and he's been using a custom board from his employer.

Using a board that is supported by upstream OpenWRT/LEDE would not be a bad idea though.

mattsm, 2017-03-04:
I disagree, it has one ethernet port and that is only 10/100. And there is no open source component for many items on that board e.g. wifi.

I'd suggest some MIPs based QCA board or ar71xx, but I'm a bit biased here. Any of the cheap boards that are at least already well supported in OpenWrt that have two real ethernet ports, a builtin switch, and decent wifi with ideal an open source driver available.

I know this is old now, but I got this for $30 and it's a much better reference platform than an rpi3.

https://slickdeals.net/f/9058015-trendnet-tew-823dru-dual-band-wireless-ac1750-gigabit-router-29-95-with-free-shipping?p=88921359

-M

madscientist42, 2018-01-01:
You can always get a GL-Inet based micro board for ~$35.   Using the PI3 isn't bad.  Just because it's got a "slow" Ethernet port means little.  One could also contemplate one of the Allwinner based ARM boards as well.

Kind of moot, though...it's not building clean on some of the autotools based packagings.  Trying to figure that out as I work my way to a Pi3 and some of the Linux-sunxi based systems.

madscientist42, 2018-01-01:
(And no open source components for the WiFi?   Odd...   It's in the kernel and only needs proprietary firmware...like most of the other drivers...)

cshoredaniel, 2018-01-15:
@mattsm the problem with the ar71xx boards is that Openwrt has a huge technical debt of kernel patches that have never been upstreamed so the only way to get most of them to work is to create a custom kernel tree with openwrt patches imported.  It looks like this is being worked on though (generic kernel stuff has been dived into 'backport', 'pending', and 'hack' patches so far).

mattsm, 2018-01-16:
rpi3 have the same problem of custom tree's. I've thought a little about a kernel bbappends that brings in the right patches that OpenWrt needs to cover this...

cshoredaniel, 2018-01-16:
@mattsm it's on my radar actually, but not sure where it will go.

cshoredaniel, 2018-01-17:
@mattsm I have a copy of automatic imports into linux-yocto @ https://github.com/cshoredaniel/cshore-linux-yocto-4.4-openwrt and https://github.com/cshoredaniel/cshore-linux-yocto-4.9-openwrt

cshoredaniel, 2018-01-20:
@mattsm In case you were actually looking at those repos, I moved them to https://github.com/cshore-yocto-4-4/linux-yocto-4.4-openwrt and https://github.com/cshore-yocto-4-9/linux-yocto-4.9-openwrt
because I made them forks of torvalds/linux and added the Yocto and Openwrt differences (so that I wouldn't be duplicating umpteen billion commits already on github).  (Actually still working on pushing 4.9 as I write, but by the time you see this it should be there).