kraj/meta-openwrt #17: Which init should be used to started the rootfs?

tonyho, 2017-03-15:
I'd like to booting the rootfs with some service start, like luci and other. And I find that there're 3 init in the openwrt-base-image:
tmp/work/imx6ul_var_dart-poky-linux-gnueabi/openwrt-base-image/1.0-r0/rootfs/sbin/init
tmp/work/imx6ul_var_dart-poky-linux-gnueabi/openwrt-base-image/1.0-r0/rootfs/usr/sbin/init
tmp/work/imx6ul_var_dart-poky-linux-gnueabi/openwrt-base-image/1.0-r0/rootfs/init

And the last one just starts a shell, and do nothing. So how can I start the luci and other openWRT router things to get it work?

kraj, 2018-01-17:
usually /sbin/init is the right one

cshoredaniel, 2018-01-19:
@tonyho Now that the initab is right for arm with procd /sbin/init (which is automatically used by runqemu) is the right one.  Also you may want to check the updated README and post- #40 pull as the procd init it now working properly on arm and x86 (at least).

kraj, 2018-01-19:
seems it fixed. I will close it.

closed 2018-01-19 by kraj