kraj/meta-openwrt #23: fix README.md instructions to get working openwrt system

milksob, 2017-09-10:
Following current instruction from README doesn't result in working image (I tried qemux86) with web interface and other features.

"libubox-examples" gives nothing
"juci" -- error while build

kraj, 2018-01-17:
Please send a pull and I will accept it

milksob, 2018-01-17:
I'll be happy to send pull request as soon as I get openwrt from this
layers myself ;-)

closed 2018-01-19 by kraj

cshoredaniel, 2018-01-19:
@milksob You might want to try with a fresh pull and reading the updated README!

cshoredaniel, 2018-01-23:
@milksob if you try using oe-core master or rocko, you may need to revert oe-core commit 7267e7c000c76c44d09835d4cd2bc485b6a39a2a to be able to add layers; there seems to be some funkiness with bitbake add-layers with that commit, but it seems to be not to be an issue for all users, but some do see it.

milksob, 2018-01-23:
thx! I'll check as soon as I get some time