kraj/meta-openwrt #27: libubox.so incorrectly packaged

cshoredaniel, 2018-01-15:
I noticed that libubox.so is being packaged in libubox-dev instead of libubox proper and that somehow in my packaging of ubox (not the lib) I was able to package and install libubox.so which allowed the base image to work, but would fail for using the various packages without the ubox package (which is likely not going to be used outside a openwrt-base-image type scenario).

I'm working on this....

kraj, 2018-01-17:
yes thats a bug, OE defaults are to assume unversioned .so to be symlinks pointing to proper versioned .so which is not case here. So you have to let the system know that via recipe by redefining 
```
SOLIBS = ".so"
FILES_SOLIBSDEV = ""
 ```

cshoredaniel, 2018-01-17:
Ah, didn't know about 'SOLIBS = ".so"' part.  Need to update a couple of recipes with that.  (I'm a relative newbie to OpenEmbedded, so I'm still learning).

closed 2018-01-19 by kraj