kraj/meta-openwrt #36: luci overview and realtime graphs fail to update

cshoredaniel, 2018-01-18:
Probably a path issue or issues.  luci overview has some information but not all that is supposed to be there (and uhttpd spits lua errors to syslog about the overview rendering).

kraj, 2018-01-19:
Can you add the error msgs here

cshoredaniel, 2018-01-19:
will do

cshoredaniel, 2018-01-20:
For the overview:
`Sat Jan 20 15:14:39 2018 daemon.err uhttpd[439]: A runtime error occured: [string "/usr/lib/lua/5.1/luci/view/admin_status/ind..."]:34: bad argument #1 to 'tonumber' (string expected, got nil)
Sat Jan 20 15:14:39 2018 daemon.err uhttpd[439]: stack traceback:
Sat Jan 20 15:14:39 2018 daemon.err uhttpd[439]:        [C]: in function 'assert'
Sat Jan 20 15:14:39 2018 daemon.err uhttpd[439]:        /usr/lib/lua/5.1/luci/dispatcher.lua:460: in function 'dispatch'
Sat Jan 20 15:14:39 2018 daemon.err uhttpd[439]:        /usr/lib/lua/5.1/luci/dispatcher.lua:141: in function </usr/lib/lua/5.1/luci/dispatcher.lua:140>
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: /usr/lib/lua/5.1/luci/dispatcher.lua:460: Failed to execute function dispatcher target for entry '/'.
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: The called action terminated with an exception:
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: /usr/lib/lua/5.1/luci/dispatcher.lua:460: Failed to execute firstchild dispatcher target for entry '/admin'.
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: The called action terminated with an exception:
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: /usr/lib/lua/5.1/luci/dispatcher.lua:460: Failed to execute function dispatcher target for entry '/admin/status'.
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: The called action terminated with an exception:
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: /usr/lib/lua/5.1/luci/dispatcher.lua:460: Failed to execute template dispatcher target for entry '/admin/status/overview'.
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: The called action terminated with an exception:
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: /usr/lib/lua/5.1/luci/template.lua:97: Failed to execute template 'admin_status/index'.
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: A runtime error occured: [string "/usr/lib/lua/5.1/luci/view/admin_status/ind..."]:34: bad argument #1 to 'tonumber' (string expected, got nil)
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]: stack traceback:
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]:        [C]: in function 'assert'
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]:        /usr/lib/lua/5.1/luci/dispatcher.lua:460: in function 'dispatch'
Sat Jan 20 15:14:44 2018 daemon.err uhttpd[439]:        /usr/lib/lua/5.1/luci/dispatcher.lua:141: in function </usr/lib/lua/5.1/luci/dispatcher.lua:140>
`

cshoredaniel, 2018-01-20:
Attempting to view the system logs:
````Sat Jan 20 15:17:59 2018 daemon.err uhttpd[439]: logread: can't find syslogd buffer: No such file or directory````

cshoredaniel, 2018-01-20:
The realtime graphs don't throw an error but I did notice that in the OE build directory (that is the directory from which one runs bitbake), that /bin/luci-bwc is created, which is supposed to on the target, and I know that file is where luci gets the stats.

closed 2018-01-24 by kraj