kraj/meta-openwrt #41: Should recipes-tweaks actually exist?

cshoredaniel, 2018-01-19:
.bbappends aren't usually split out as recipe-tweaks are they?  Are they not simply put in place according the recipe (e.g. recipes-core, recipes-networking, etc)?

kraj, 2018-01-19:
I dont have hard feelings about it. recipe-tweaks usually was added to house all bbappends for recipes from other layers. This would make it easy to exclude it from parsing using BBMASK if needed. In cases someone was including meta-openwrt but was not really using it. It also means we keep track of recipes we modified since that number should be kept minimal. but its ok if we use some override e.g. openwrt to make all these changes in bbappend.

cshoredaniel, 2018-01-19:
Ok, I didn't know about the BBMASK thing.  I actually included a PACKAGECONFIG check to base-files to omit the openwrt stuff, but I guess that's not totally necessary since BBMASK can omit it.  (I was thinking of e.g. someone wanting to use procd but not the whole stack).

kraj, 2018-01-19:
yes, we have thought about moving procd to oe-core but how openwrt sources are organised, I do not know how much independent component it is

cshoredaniel, 2018-01-19:
You might want to take a look at what I've done; there is some tweaking to be done, but I think I've identified the dependencies (e.g. ubus, libubox) and removed the dependency on modified base-files (but do pull from base-files package in openwrt as part of build).  Some of this would need to be modified to used standalone in oe-core.  I've successfully build core-image-minimal with procd as init_manager/dev_manager but no other openwrt (except ubus, libubox) if that helps.

cshoredaniel, 2018-01-22:
Looks like recipes-tweaks makes sense, so closing.  I don't have a definite drive to move things out of tweaks (and it makes it clear the sources are from other layers).

closed 2018-01-22 by cshoredaniel