kraj/meta-openwrt #77: Upgrading OpenWRT packages to latest git revision or AUTOREV

parthitce, 2018-04-17:
All the OpenWRT software source is currently referring to older revision in git. I don't see any reasons for not upgrading it to latest git revision or AUTOREV.

Also there is no stable release or branching OpenWRT project as of now. This issue is created to discuss the possibility to upgrade the OpenWRT packages and to get more thoughts about it.

cshoredaniel, 2018-04-17:
Hi,

Glad to see someone using this; I've been dealing with personal stuff 
for some months now and haven't been working on this as I'd like.

> All the OpenWRT software source is currently referring to older revision 
> in git. I don't see any reasons for not upgrading it to latest git 
> revision or AUTOREV.

Actually OpenWRT was forked as LEDE and the the current git revisions 
are based on LEDE 17.01.4 which is the current stable point release. 
There is a new release coming out RSN (Real Soon Now), possibly this 
month or latest next at which point it would make sense to update before 
branching for OE Sumo (IMO).

After Sumo I think it would make sense to have versioned recipes and 
.inc's so that there are recipes corresponding to the stable OpenWrt and 
'real' git version that corresponds to OpenWRT master.

I think the remerge of LEDE and OpenWrt looks stalled, and so the web 
presence is a little messed up at the moment, best way to find out what 
is going on is to subscribe to both lede-dev list and openwrt-devel list.

Regards,

Daniel

parthitce, 2018-04-17:
Hi,

> Hi,
> 
> Glad to see someone using this; I've been dealing with personal stuff
> for some months now and haven't been working on this as I'd like.

Thanks. Although am not using the meta layer completely, I have ported 
few core applications to Linux custom project requirements.

> Actually OpenWRT was forked as LEDE and the the current git revisions
> are based on LEDE 17.01.4 which is the current stable point release.
> There is a new release coming out RSN (Real Soon Now), [...]

Sure, that make sense. Do you think keeping this issue open until next 
LEDE release is reasonable?

> After Sumo I think it would make sense to have versioned recipes and
> .inc's so that there are recipes corresponding to the stable OpenWrt and
> 'real' git version that corresponds to OpenWRT master. [...]

Will try to follow up.

> 
> Regards,
> 
> Daniel
> 
>>
>> Also there is no stable release or branching OpenWRT project as of now.
>> This issue is created to discuss the possibility to upgrade the OpenWRT
>> packages and to get more thoughts about it.

kraj, 2018-04-17:
ideally we want to have a setup where we can set a global SRCREV  = AUTOREV and build everything from latest master always and main recipes are always locked to a well tested ( stable ) version

bachp, 2018-04-18:
I think this could be done similar to poky-bleeding using a file like: https://github.com/bradfa/meta-yocto/blob/master/meta-yocto/conf/distro/include/poky-floating-revisions.inc