kraj/meta-openwrt #79: Wi-Fi not working in OpenWRT Raspberrypi3 - wlan0 interface is not up

satishnaidu, 2018-05-08:
**Openwrt on RaspberryPi wifi not working - wlan0 interface is not up**

conf/local.conf
---------------
MACHINE ??= "raspberrypi3"
INHERIT += " openwrt-distro-defaults "
DISTRO_FEATURES_append = " virtualization wifi"
MACHINE_EXTRA_RRECOMMENDS += "kernel-modules linux-firmware-bcm43430"
TCLIBC = "musl"
IMAGE_INSTALL_append = " initscripts hostapd linux-firmware-bcm43430 bridge-utils netifd bash openssh kernel-modules docker iptables wireless-tools dhcp-client  crda iw linux-firmware-ralink linux-firmware-rtl8192ce linux-firmware-rtl8192cu linux-firmware-rtl8192su "
EXTRA_IMAGE_FEATURES ?= "debug-tweaks package-management"

conf/bblayers.conf
------------------
BBLAYERS ?= " \
  /home/ubuntu/yocto/latest/openembedded-core/meta \
  /home/ubuntu/yocto/latest/openembedded-core/meta-openembedded/meta-oe \
  /home/ubuntu/yocto/latest/openembedded-core/meta-openembedded/meta-python \
  /home/ubuntu/yocto/latest/openembedded-core/meta-openembedded/meta-networking \
  /home/ubuntu/yocto/latest/openembedded-core/meta-openembedded/meta-filesystems \
  /home/ubuntu/yocto/latest/openembedded-core/meta-virtualization \
  /home/ubuntu/yocto/latest/openembedded-core/meta-openwrt \
  /home/ubuntu/yocto/latest/openembedded-core/meta-raspberrypi \
  
  
/etc/config/wireless
--------------------
config wifi-device 'radio0'
		option type 'mac80211'
		option channel '11'
		option hwmode '11g'
		option path '/platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
		option htmode 'HT20'
		option disabled '0'
		
config wifi-iface 'default-radio0'
		option device 'radio0'
		option network 'lan'
		option mode    'ap'
		option ssid    'WifiName'
		option encryption 'none'
		
/etc/config/network
--------------------
config interface 'loopback'
		option ifname 'lo'
		option proto 'static'
		option ipaddr '127.0.0.1'
		option netmask '255.0.0.0'
		
config interface 'lan'
		option ifname 'eth0'
		option proto 'static'
		option type 'bridge'
		option ipaddr '192.168.1.1'
		option netmask '255.255.255.0'
		option ip6assign '60'

config interface 'wan'
		option ifname 'wlan0'
		option proto  'dhcp'

Kernel config fragments: kconfig.cfg
----------------------------------------
CONFIG_RTL8192CE=y
CONFIG_RTL8192C_COMMON=y
CONFIG_RTLWIFI=y
CONFIG_MAC80211=y
CONFIG_CFG80211=y

CONFIG_WIRELESS_EXT=y
CONFIG_WEXT_CORE=y
CONFIG_WEXT_PROC=y
CONFIG_WEXT_SPY=y
CONFIG_WEXT_PRIV=y
CONFIG_CFG80211_WEXT=y
CONFIG_LIB80211=y
CONFIG_LIB80211_CRYPT_WEP=y
CONFIG_LIB80211_CRYPT_CCMP=y
CONFIG_LIB80211_CRYPT_TKIP=y
# CONFIG_LIB80211_DEBUG is not set
CONFIG_HOSTAP=y
# CONFIG_HOSTAP_FIRMWARE is not set

CONFIG_BRCMUTIL=y
# CONFIG_BRCMSMAC is not set
CONFIG_BRCMFMAC=y
CONFIG_BRCMFMAC_PROTO_BCDC=y
CONFIG_BRCMFMAC_SDIO=y
CONFIG_BRCMFMAC_USB=y
CONFIG_BRCMFMAC_PROTO_MSGBUF=y
CONFIG_WEXT_CORE=y
CONFIG_WEXT_PROC=y

I am not seeing any errors in "dmesg" or "logread", but the wifi is not working on OpenWRT.
@kraj  Could you please let me know, what are all the configurations required to bring up the wifi on OpenWRT.

cshoredaniel, 2018-05-09:
Are you including netifd and using it or are you using hostapd directly?

What processes are running?

How are you trying to bring up the wireless?  This layer is relatively 
experimental, so things like wireless may not 'automagically' work. 
Part of that may be because regular OpenWrt uses a modified 
wireless-next tree, hostapd, etc (though thing should still mostly work 
AIUI).

Is your wireless supported on OE without OpenWrt?  I seem to recall some 
discussion around the Pi 3 wireless module support.

Regards,

Daniel

kraj, 2018-05-09:
if you use latest master meta-raspberrypi layer it uses right firmware blobs to bringup on-board wifi
I have not tried to bring up wifi on pi3 myself with openwrt layer but it works fine with OE in general

sandeepyedire, 2018-05-09:
@cshoredaniel 
We are including netifd and using it.  
when we try apply the wifi config and try bringing up wlan0 is shown up in ifconfig -a but wlan0 link is not ready.  Please suggest how to get it work on yocto with openwrt layer.
Below is our configuration to bring up wifi on opewrt platform.
In file /etc/config/wireless:
config wifi-iface
    option ssid 'linksys'  # the wifi name its connecting too
    option device 'radio0' # the radio we are using to connect
    option mode 'sta' # client mode
    option key 'password' # wifi password
    option encryption 'psk2+ccmp' # encryption level
    option network 'wwan' # interface name (used for dhcp and firewall zone etc)
    option bssid 'xx:xx:xx:xx:xx:xx'
In file /etc/config/network
config interface 'wwan'
    option proto 'dhcp'
In file /etc/config/firewall
config zone
    option name 'wan'
    option input 'REJECT'
    option output 'ACCEPT'
    option forward 'REJECT'
    option masq '1'
    option mtu_fix '1'
    option network 'wan wwan'

#/etc/init.d/firewall restart # restarts firewall
#/etc/init.d/network restart # restarts network
#wifi    #restarts wifi**

**lsmod on the yocto build:**
root@LEDE:/# lsmod
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
bridge                122880  0
cfg80211              573440  1 brcmfmac
evdev                  24576  0
fixed                  16384  0
ip_tables              24576  3 iptable_nat,iptable_mangle,iptable_filter
ip6_tables             24576  2 ip6table_mangle,ip6table_filter
ip6table_filter        16384  0
ip6table_mangle        16384  0
ipt_MASQUERADE         16384  1
ipt_REJECT             16384  2
iptable_filter         16384  1
iptable_mangle         16384  0
iptable_nat            16384  1
ipv6                  438272 37 nf_conntrack_ipv6,nf_defrag_ipv6,nf_nat_masquerade_ipv6,ip6table_mangle,bridge,[permanent]
llc                    16384  2 bridge,stp
nf_conntrack          126976 14 nf_nat_ipv4,nf_conntrack_netlink,nf_conntrack_ipv4,ipt_MASQUERADE,xt_state,xt_nat,xt_conntrack,xt_REDIRECT,xt_NETMAP,xt_CT,nf_nat_masquerade_ipv4,nf_conntrack_ipv6,nf_nat_masquerade_ipv6,nf_nat
nf_conntrack_ipv4      16384 10
nf_conntrack_ipv6      16384  0
nf_conntrack_netlink   36864  0
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
nf_defrag_ipv6         24576  1 nf_conntrack_ipv6
nf_log_common          16384  1 nf_log_ipv4
nf_log_ipv4            16384  0
nf_nat                 28672  6 nf_nat_ipv4,xt_nat,xt_NETMAP,nf_nat_redirect,nf_nat_masquerade_ipv4,nf_nat_masquerade_ipv6
nf_nat_ipv4            16384  1 iptable_nat
nf_nat_masquerade_ipv4   16384  1 ipt_MASQUERADE
nf_nat_masquerade_ipv6   16384  0
nf_nat_redirect        16384  1 xt_REDIRECT
nf_reject_ipv4         16384  1 ipt_REJECT
nfnetlink              16384  1 nf_conntrack_netlink
rfkill                 32768  1 cfg80211
stp                    16384  1 bridge
uio                    20480  1 uio_pdrv_genirq
uio_pdrv_genirq        16384  0
x_tables               32768 23 ipt_MASQUERADE,xt_time,xt_tcpudp,xt_tcpmss,xt_state,xt_nat,xt_multiport,xt_mark,xt_mac,xt_limit,xt_conntrack,xt_comment,xt_REDIRECT,xt_NETMAP,xt_LOG,xt_CT,iptable_mangle,iptable_filter,ip_tables,ipt_REJECT,ip6table_mangle,ip6table_filter,ip6_tables
xt_CT                  16384  0
xt_LOG                 16384  0
xt_NETMAP              16384  0
xt_REDIRECT            16384  0
xt_comment             16384 50
xt_conntrack           16384  8
xt_limit               16384  1
xt_mac                 16384  0
xt_mark                16384  0
xt_multiport           16384  0
xt_nat                 16384  0
xt_state               16384  0
xt_tcpmss              16384  0
xt_tcpudp              16384  4
xt_time                16384

cshoredaniel, 2018-05-09:
> @cshoredaniel <https://github.com/cshoredaniel>
> We are including netifd and using it.
> when we try apply the wifi config and try bringing up wlan0 is shown up
> in ifconfig -a but wlan0 link is not [...]

Ah, I though you were doing AP mode.  Do you have wpa_supplicant?  (I'm 
not sure if that's enough or if a recipe doing OpenWRT-like multi-call 
wapd is necessary for client mode).

Does the client attempt to make a connection but fail (i.e. when you 
watch wireless traffic)?  Do you have busybox's udhcpcd included? 
(netifd doesn't grok dhcp-client from ISC).

Regards,

Daniel

satishnaidu, 2018-05-09:
Hi @cshoredaniel ,

Please find the below observations:
Do you have wpa_supplicant?  we are not using any wpa_supplicant file explicitly in the conf/local.conf file.
OpenWRT-like multi-call wapd is necessary for client mode: I tried to include wpad in OpenWRT through Yocto, but there is no recipe available. ( I assume we don't need this module, because we are not doing anything complex just simple wifi client)

Does the client attempt to make a connection but fail : In the logread we are not seeing any errors, the connection itself is not established so not getting any errors. the netifd is not eveing loading the wan interface when we specify ifname as wlan0.

But when we configure ifname as eth0 instead of wlan0 in , /etc/config/network 

config interface 'wan'
option ifname 'etho0'
option proto 'dhcp'

In the logread, netifd is trying to loading the wan interface but not getting any ip address.

Wed May  9 21:29:26 2018 daemon.notice netifd: Network device 'eth0' link is up
Wed May  9 21:29:26 2018 daemon.notice netifd: Interface 'wan' has link connectivity 
Wed May  9 21:29:26 2018 daemon.notice netifd: Interface 'wan' is setting up now
Wed May  9 21:29:26 2018 daemon.notice netifd: wan (1970): udhcpc: started, v1.27.2
Wed May  9 21:29:26 2018 daemon.notice netifd: wan (1970): udhcpc: sending discover
Wed May  9 21:29:26 2018 daemon.notice netifd: bridge 'br-lan' link is up
Wed May  9 21:29:26 2018 daemon.notice netifd: Interface 'lan' has link connectivity 
Wed May  9 21:29:27 2018 daemon.info dnsmasq[569]: read /etc/hosts - 6 addresses
Wed May  9 21:29:27 2018 daemon.info dnsmasq[569]: read /tmp/hosts/odhcpd - 3 addresses
Wed May  9 21:29:27 2018 daemon.info dnsmasq[569]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Wed May  9 21:29:28 2018 daemon.info odhcpd[450]: Using a RA lifetime of 0 seconds on br-lan
Wed May  9 21:29:29 2018 daemon.notice netifd: wan (1970): udhcpc: sending discover
Wed May  9 21:29:32 2018 daemon.notice netifd: wan (1970): udhcpc: sending discover
Wed May  9 21:29:35 2018 daemon.notice netifd: wan (1970): udhcpc: sending discover
Wed May  9 21:29:39 2018 daemon.notice netifd: wan (1970): udhcpc: sending discover
Wed May  9 21:29:41 2018 kern.info kernel: [16096.059780] brcmfmac: power management disabled
Wed May  9 21:29:41 2018 kern.info kernel: [16096.065571] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed May  9 21:29:42 2018 daemon.notice netifd: wan (1970): udhcpc: sending discover
Wed May  9 21:29:45 2018 daemon.notice netifd: wan (1970): udhcpc: sending discover
Wed May  9 21:29:48 2018 daemon.notice netifd: wan (1970): udhcpc: sending discover
Wed May  9 21:29:51 2018 daemon.notice netifd: wan (1970): udhcpc: sending discover

udhcp: keep sendin discover and doesn't assigning any ip address.

Do you have busybox's udhcpcd included: udhcpd is already included in the image.

Any pointers, to resolve this wifi issue is highly appreciated?

Thanks,
Satish Kumar Andey

cshoredaniel, 2018-05-11:
> Hi @cshoredaniel <https://github.com/cshoredaniel> ,
>
> Please find the below observations:
> Do you have wpa_supplicant? we are not using any wpa_supplicant file
> explicitly in the conf/local.conf [...]
AIUI wpa_supplicant will work with netifd too.  You need at least one of 
wpa_supplicant or wpad to do wireless with netifd.

> ( I assume we don't need this module, because we are not doing anything
> complex just simple wifi client)
>
> Does the client attempt to make a connection but fail : In the logread
> we are not seeing [...]

If you don't have a wired connection then that is understandable.  This 
should, however, work with a wired connection.

With netifd eth0 will be associated with the physical ethernet port.

satishnaidu, 2018-05-15:
Hi @cshoredaniel,

As per your suggestion, I have added wpa_supplicant in local.conf file able to bring up the wifi on  OpenWRT.
The steps I have followed is as follows:
1. ifconfig wlan0 up
2. wpa_passphrase 'SSID' 'Password' > /etc/wpa_supplicant.conf
3. wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant.conf
4. dhclient wlan0
or
1. iwconfig wlan0 'SSID' ( For WEP encryption)
2. dhclient wlan0

Now my question is, if we configure the wifi details through wpa_supplicant file, what is the purpose of /etc/config/wireless file.
If we want to change the wifi details through "uci command", is there a way to update this wpa_supplicant file?

iuncuim, 2018-08-27:
@cshoredaniel @kraj I think that iwinfo with this bb don't include nl80211 API. Maybe this is the problem?

cshoredaniel, 2018-08-27:
@iuncuim that would make sense...best thing to do is look at https://git.openwrt.org/openwrt/openwrt.git I guess.

FallenChromium, 2021-01-09:
As far as I understood, the issue comes from brcmfmac. Every time you restart hostapd using UCI, it grabs HT capabilities from iw list, which in turn gets it from the driver. If you set up 80211n=0 (so basically Legacy mode) it will work, because HT is then disabled. Driver reports SHORT-GI-HT40 is supported, but it can't bring up the interface with this option, because it requires falling back to mimo_bw_cap (patch available in the open). Using wl binary from cypress website and setting mimo_bw_cap manually helps. I am looking on the ways how it can be fixed now.