kraj/meta-openwrt #80: LIC_FILES_CHKSUM points to an invalid file

LeosSire, 2018-05-30:
Hello,

While compiling I recieved the below warning. Please can anyone advise?

Compiling on Debian 8. 

`
WARNING: popt-native-1.16-r3 do_fetch: Failed to fetch URL http://rpm5.org/files/popt/popt-1.16.tar.gz, attempting MIRRORS if available
WARNING: shadow-4.2.1-r0 do_fetch: Failed to fetch URL http://pkg-shadow.alioth.debian.org/releases/shadow-4.2.1.tar.xz, attempting MIRRORS if available
ERROR: base-files-3.0.14-r89 do_populate_lic: QA Issue: base-files: LIC_FILES_CHKSUM points to an invalid file: /home/user/openembedded-core/build/tmp-musl/work/qemuarm-oe-linux-musleabi/base-files/3.0.14-r89/../git/openwrt/LICENSE [license-checksum]
NOTE: Tasks Summary: Attempted 2101 tasks of which 5 didn't need to be rerun and all succeeded.

Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
`

kraj, 2018-05-30:
I think rpm5.org is down at the moment. you might be able to use

ftp://anduin.linuxfromscratch.org/BLFS/popt/popt-1.16.tar.gz

LeosSire, 2018-05-30:
Thank you Kraj. 

I tried rerunning the bitbake and it ran straight through without retrying to download and compile, and without error. 

I’m a complete noob coming across from OpenWRT. Does this mean it compiled anyway or do I have to remove temp files and recompile. 

Using qemu stalled on lack of display device. Don’t know if this is related. Not had time to look into it. 

Any advise is appreciated. 

Joe

kraj, 2018-05-30:
I think it should be ok then

kraj, 2018-07-03:
for qemu you sould run it with
```
runqemu nographic
```

this will ensure that qemu runs in shell without need of X11