kraj/meta-openwrt #83: Unable to add additional layers

cclark1974, 2018-07-03:
I am attempting to compile with added layers and I encountering errors.
 
bitbake-layers show-layers
NOTE: Starting bitbake server...
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/root/Yocto/poky/Cl-Img/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 9855 at 2018-07-03 13:26:59.125015 ---
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass

ERROR: Unable to start bitbake server
ERROR: Server log for this session (/root/Yocto/poky/Cl-Img/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 9855 at 2018-07-03 13:26:59.125015 ---
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass

My local.conf
cat conf/local.conf 
#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# There are also the following hardware board target machines included for 
# demonstration purposes:
#
#MACHINE ?= "beaglebone-yocto"
#MACHINE ?= "genericx86"
#MACHINE ?= "genericx86-64"
#MACHINE ?= "mpc8315e-rdb"
#MACHINE ?= "edgerouter"
#
# This sets the default machine to be qemux86 if no other machine is selected:
MACHINE ??= "qemux86"
[... 221 lines cut ...]

my bblayers.conf

cat conf/bblayers.conf 
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /root/Yocto/poky/meta \
  /root/Yocto/poky/meta-poky \
  /root/Yocto/poky/meta-openwrt \
  /root/Yocto/poky/meta-openembedded/meta-initramfs \
  /root/Yocto/poky/meta-openembedded/meta-networking \
  /root/Yocto/poky/meta-openembedded/meta-oe \
  /root/Yocto/poky/meta-openembedded/meta-python \
  /root/Yocto/poky/meta-sysrepo \
  /root/Yocto/poky/meta-openembedded/meta-webserver \
  /root/Yocto/poky/meta-yocto-bsp \
  "

Any other info thats needed please let me know.  Much appreciated.

kraj, 2018-07-03:
Add 

```
LAYERSERIES_COMPAT_sysrepo = "sumo"
```
in meta-sysrepo/conf/layer.conf

secondly somewhere you seem to have

inherit classes/openwrt-distro-defaults.bbclass this is not correct you only need

```
INHERIT += " openwrt-distro-defaults "
```

cclark1974, 2018-07-05:
I cleaned up the directory by removing the cache, downloads, sstate-cache and tmp directories

I searched local.conf and bblayer.conf and did not file openwrt-distro-defaules.bbclass in any area and I'm still hitting the bitbake-layers show-layers error.

[root@hostname Cl-Img]# bitbake-layers show-layers
NOTE: Starting bitbake server...
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/root/Yocto/poky/Cl-Img/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 11685 at 2018-07-05 11:56:16.136696 ---
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass

ERROR: Unable to start bitbake server
ERROR: Server log for this session (/root/Yocto/poky/Cl-Img/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 11685 at 2018-07-05 11:56:16.136696 ---
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass

[root@hostname Cl-Img]# 

my local.conf is:
`#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# There are also the following hardware board target machines included for 
# demonstration purposes:
#
#MACHINE ?= "beaglebone-yocto"
#MACHINE ?= "genericx86"
#MACHINE ?= "genericx86-64"
#MACHINE ?= "mpc8315e-rdb"
#MACHINE ?= "edgerouter"
[... 270 lines cut ...]

and my bblayers.conf is:
`# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /root/Yocto/poky/meta \
  /root/Yocto/poky/meta-poky \
  /root/Yocto/poky/meta-openwrt \
  /root/Yocto/poky/meta-openembedded/meta-initramfs \
  /root/Yocto/poky/meta-openembedded/meta-networking \
  /root/Yocto/poky/meta-openembedded/meta-oe \
  /root/Yocto/poky/meta-openembedded/meta-python \
  /root/Yocto/poky/meta-sysrepo \
  /root/Yocto/poky/meta-openembedded/meta-webserver \
  /root/Yocto/poky/meta-yocto-bsp \
  "`

I added the LAYERSERIES_COMPAT in local and still am having the issue.

Thanks

cshoredaniel, 2018-07-06:
Hi, it's hard to read your configs because they are formatted badly.  Please edit and reformat as 'code blocks' so that they get displayed verbatim instead of as if they were Markdown soure.

can you verify that BBLAYERS is actually set to what you have in bblayers.conf?  
``bitbake -e <target-image>`` (IIRC)

cshoredaniel, 2018-07-06:
I'm wondering about your BBLAYERS.CONF.  @kraj can correct me if I'm wrong, but normally bitbake lists are space-separated not newline seperated, and the backslash (\\) character is used at the end of the line so the variable is process properly.

cclark1974, 2018-07-06:
`bitbake -e openwrt-image-full
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/root/Yocto/poky/Cl-Img/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 12727 at 2018-07-06 13:28:28.629920 ---
WARNING: Layer appliance should set LAYERSERIES_COMPAT_appliance in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer sysrepo should set LAYERSERIES_COMPAT_sysrepo in its conf/layer.conf file to list the core layer names it is compatible with.
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/openwrt-distro-defaults.bbclass`

cshoredaniel, 2018-07-08:
Look like you need the sumo compat thing in the 'appliance' layer as well and missed that one?