kraj/meta-openwrt #89: Build openwrt-image-minimal failed

tuanha-viosoft, 2019-02-25:
I build the openwrt-image-minimal and get error:
build-openwrt$ TCLIBC=musl MACHINE=qemuarm bitbake openwrt-image-minimal
Parsing recipes: 100% |##########################################| Time: 0:00:22
Parsing of 2242 .bb files complete (0 cached, 2242 parsed). 3291 targets, 171 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "arm-oe-linux-musleabi"
MACHINE              = "qemuarm"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm armv5 thumb dsp"
TARGET_FPU           = "soft"
meta                 = "master:ff2c029b568f70aa9960dde04ddd207829812ea0"
meta-oe
meta-multimedia
meta-python
meta-networking      = "master-next:fbc644835875df5ee74bbafc295b4b13f347ec51"
meta-openwrt         = "master:96a8e2de1ea631aaea320ffd268c6b194c7c77e1"

Initialising tasks: 100% |#######################################| Time: 0:00:01
Sstate summary: Wanted 594 Found 0 Missed 594 Current 0 (0% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: alsa-lib-native-1.1.6-r0 do_fetch: Failed to fetch URL ftp://ftp.alsa-project.org/pub/lib/alsa-lib-1.1.6.tar.bz2, attempting MIRRORS if available      WARNING: netbase-1_5.5-r0 do_fetch: Failed to fetch URL http://ftp.debian.org/debian/pool/main/n/netbase/netbase_5.5.tar.xz, attempting MIRRORS if available    ERROR: xtables-addons-3.2-r0 do_compile: oe_runmake failed                      ERROR: xtables-addons-3.2-r0 do_compile: Function failed: do_compile (log file is located at /home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/temp/log.do_compile.9482)     
...
| /home/ubuntu/build-openwrt/tmp-musl/work-shared/qemuarm/kernel-source/Makefile:1508: recipe for target '_module_/home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/build/../xtables-addons-3.2/extensions' failed
| make[4]: *** [_module_/home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/build/../xtables-addons-3.2/extensions] Error 2
| make[4]: Leaving directory '/home/ubuntu/build-openwrt/tmp-musl/work-shared/qemuarm/kernel-build-artifacts'
| Makefile:146: recipe for target 'sub-make' failed
| make[3]: *** [sub-make] Error 2
| make[3]: Leaving directory '/home/ubuntu/build-openwrt/tmp-musl/work-shared/qemuarm/kernel-source'
| Makefile:459: recipe for target 'modules' failed
| make[2]: *** [modules] Error 2
| make[2]: Leaving directory '/home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/build/extensions'
| ERROR: oe_runmake failed
| Makefile:492: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/build'
| Makefile:377: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/temp/log.do_compile.9482)
ERROR: Task (/home/ubuntu/riscv-yocto/meta-openwrt/recipes-core/xtables-addons/xtables-addons_3.2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1671 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ubuntu/riscv-yocto/meta-openwrt/recipes-core/xtables-addons/xtables-addons_3.2.bb:do_compile
Summary: There were 2 WARNING messages shown.

kraj, 2019-02-26:
Can you post the error message where it indicates what went wrong ? it should be inside log.do_compile

tuanha-viosoft, 2019-02-26:
/home/ubuntu/build-openwrt/tmp-musl/work-shared/qemuarm/kernel-source/Makefile:1508: recipe for target '_module_/home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/build/../xtables-addons-3.2/extensions' failed
make[4]: *** [_module_/home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/build/../xtables-addons-3.2/extensions] Error 2
make[4]: Leaving directory '/home/ubuntu/build-openwrt/tmp-musl/work-shared/qemuarm/kernel-build-artifacts'
Makefile:146: recipe for target 'sub-make' failed
make[3]: *** [sub-make] Error 2
make[3]: Leaving directory '/home/ubuntu/build-openwrt/tmp-musl/work-shared/qemuarm/kernel-source'
Makefile:459: recipe for target 'modules' failed
make[2]: *** [modules] Error 2
make[2]: Leaving directory '/home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/build/extensions'
ERROR: oe_runmake failed
Makefile:492: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/ubuntu/build-openwrt/tmp-musl/work/qemuarm-oe-linux-musleabi/xtables-addons/3.2-r0/build'
Makefile:377: recipe for target 'all' failed
make: *** [all] Error 2
WARNING: exit code 1 from a shell command.
[log.do_compile.txt](https://github.com/kraj/meta-openwrt/files/2903708/log.do_compile.txt)

tuanha-viosoft, 2019-02-28:
I made a patch here:
https://patchwork.openembedded.org/patch/159149/

kraj, 2019-02-28:
Thanks @tuanha-viosoft I would prefer a github pull request here

closed 2019-03-01 by tuanha-viosoft