kraj/meta-openwrt #92: openwrt-image-*: mv command not working

tuanha-viosoft, 2019-03-01:
I tried "mv" command on openwrt-image-minimal, base and full, it doesn't work.
`root@192:~# mv a b`
`mv: can't rename 'a': Function not implemented`

kraj, 2019-03-07:
what arch are you building for ? May be strace this mv cmd and see if there is some syscall not implemented.

tuanha-viosoft, 2019-03-07:
> what arch are you building for ? May be strace this mv cmd and see if there is some syscall not implemented.

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "riscv64-oe-linux-musl"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "master:c31bcb0555ae77fcb59c4d0798ec66bb27f63dc2"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "master-next:90956db4f410b162acb1468b93ebfc86786c4816"
meta-openwrt         = "master:42e95362f24deeaefaa36b374125e21e08fdd6fb"
meta-riscv           = "work:5b35295781cdfd255ef16b04073f182feffe5a0e"

I get the meta-riscv from here:
https://github.com/riscv/meta-riscv

kraj, 2019-03-07:
This seems to be risc-v specific, can you try same cmd on qemuarm ?

tuanha-viosoft, 2019-03-08:
I tried, it seems that only mv command didn't work. Of course, if I replaced busybox by coreutils, mv command worked.

kraj, 2019-03-08:
I think it could be related to musl+riscv combo, I would like to see if it happens with other arches as well likely qemuarm