kraj/meta-openwrt #98: 600-shared-libext.patch does not apply to iptables 1.8.4

zeldin, 2020-12-04:
The bbappend file `meta-openwrt/recipes-tweaks/iptables/iptables_1.8.%.bbappend` adds `600-shared-libext.patch` to any iptables version matching 1.8.x.  But it does not apply to 1.8.4, which is the one used by OE dunfell.

```
ERROR: iptables-1.8.4-r0 do_patch: Command Error: 'quilt --quiltrc /home/marcus/COTM/Mercury_SW/yocto/build/tmp-angstrom-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/iptables/1.8.4-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch 600-shared-libext.patch
patching file extensions/GNUmakefile.in
patching file iptables/Makefile.am
Hunk #1 FAILED at 8.
Hunk #2 FAILED at 45.
2 out of 2 hunks FAILED -- rejects in file iptables/Makefile.am
Patch 600-shared-libext.patch does not apply (enforce with -f)
ERROR: Logfile of failure stored in: /home/marcus/COTM/Mercury_SW/yocto/build/tmp-angstrom-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/iptables/1.8.4-r0/temp/log.do_patch.3124
ERROR: Task (/home/marcus/COTM/Mercury_SW/yocto/layers/openembedded-core/meta/recipes-extended/iptables/iptables_1.8.4.bb:do_patch) failed with exit code '1'
```

kraj, 2020-12-04:
Thanks @zeldin for your report. I will take a look.

kraj, 2020-12-05:
Fixed now

closed 2020-12-05 by kraj