kraj/meta-openwrt #104: State of the luci integration

macpijan, 2021-02-05:
Hello. Is there any information on what has been proved to work in terms of the luci integration? I'm trying a build for the x86 hardware target and some of the luci features are not functioning corectly - like the resources usage in the first tab or the missing wireless network options (like wifi channels configuration or correct wifi status). 

I'm using dunfell currently, but trying master as well. 

Which target is known to work best with this meta layer? 

Anyone had similar issues with luci interface? Thank you.

katyo, 2021-03-23:
Currently it lacks packages for many luci apps.

kraj, 2021-03-23:
Please add some details and if you can cook patches even better

macpijan, 2021-03-23:
We have been experimenting with this layer a little bit, but we gave up for a moment due to not enough quality/feature set and resources.

We did end up implementing some recipes and patches. I have opened a dirty MR with those changes: https://github.com/kraj/meta-openwrt/pull/113/commits
Please take a look if any of those commits seem useful. We could potentially support with some rebase or testing