kraj/meta-openwrt #131: No recipe in default for busybox_1.34

frebour1, 2022-04-11:
Following wiki , building openwrt-image-minimal I got :

$ TCLIBC=musl MACHINE=qemuarm bitbake openwrt-image-base
Loading cache: 100% |#########################################################################################| Time: 0:00:01
Loaded 3854 entries from dependency cache.
**ERROR: No recipes in default available for:
  /mnt/yocto/openembedded-core/meta-openwrt/recipes-tweaks/busybox/busybox_1.34.%.bbappend**

coolmaxter, 2022-05-11:
Hi, did you solve the problem? I just got the same issue here.

frebour1, 2022-05-13:
I have delete following files :
recipes-tweaks/busybox/busybox_1.34.%.bbappend

kmwegger, 2022-07-07:
The best way to solve this is to add the bbappend to `BBMASK` in your `local.conf` or similar.
```
BBMASK = "recipes-tweaks/busybox/busybox_1.34.%.bbappend"
```

Using `BBMASK` you can use the layer normally, without forking or manually deleting files.

PR #138 (2023-05-05): fix build with the openembedded master branch
Fixes: #131
Fixes: #135 
Fixes: #137
Fixes: #123
commit ba137ce6 (2023-05-05): busybox: upgrade 1.34.x -> 1.36.x

Fixes: #131,#137
closed 2023-05-05 by kraj