kraj/meta-openwrt #140: general question - Is there anyway to add snort3 in this?

RNGesus-exe, 2024-01-01:
I want to enable NIDS (snort3) in meta-openwrt, Can someone explain how to add it.

kraj, 2024-01-02:
I think you need to first add recipe for snort3, which might be doable easily by looking at some other app recipes in meta-openwrt. e.g `./recipes-support/lua-socket/lua-socket_git.bb`

RNGesus-exe, 2024-01-05:
I was wondering, why isn't Snort3 already a part of the community-supported packages in meta-openwrt? Is there a specific reason it hasn't been included so far?

kraj, 2024-01-05:
> I was wondering, why isn't Snort3 already a part of the community-supported packages in meta-openwrt? Is there a specific reason it hasn't been included so far?

right, no one added it so far.

kraj, 2024-02-13:
If the recipe depends on meta-openwrt then send a PR here

RNGesus-exe, 2024-02-13:
I sent a PR of the recipe to meta-openembedded

RNGesus-exe, 2024-03-12:
Closed this issue as snort3 has been added to meta-openembedded master branch with this [PR](https://github.com/openembedded/meta-openembedded/pull/789)

closed 2024-03-12 by RNGesus-exe