kraj/meta-openwrt #142: Building Yocto/Meta-OpenWrt image for a custom Hardware

MehrozMustafa, 2024-02-19:
I am building an image for my router (Linksys WRT1900 AC v1) and I'm having trouble doing so.
I know I have to write the correct Target Machine in the MACHINE: variable in the local.conf file, but it's not working like its supposed to.
`#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# This sets the default machine to be qemux86-64 if no other machine is selected:

MACHINE ??= "qemux86-64"
#MACHINE ?= "mvebu-cortexa9"
#MACHINE ?= "marvell-armada"
#MACHINE ?= "marvell-armada-3720-db"
#TARGET ?= "mvebu"
#SUBTARGET ?= "cortexa9"

INHERIT += " openwrt-distro-defaults"
IMAGE_FSTYPES += "iso"`

Tried a bunch of values for the Target Architecture but they all failed.
Here's the link for my target machine along with its details:
https://openwrt.org/toh/linksys/wrt1900ac#hardware

MehrozMustafa, 2024-02-19:
I have commented the target architectures I tried for then kept the qemux86-64 one for building.

MehrozMustafa, 2024-02-20:
Update: I've tried all of these and then also qemuarm64 (the default ones) to check if it works on my router but it failed.
```
#MACHINE ?= "qemuarm"

MACHINE = "qemuarm64"
TCLIBC = "musl"

#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# This sets the default machine to be qemux86-64 if no other machine is selected:

#MACHINE ??= "qemux86-64"

#MACHINE = "qemuarm64"
#TCLIBC = "musl"
#MACHINE = "marvell-armada-385"
#MACHINE = "marvell-armada-37x"
#MACHINE = "marvell-armada-385-linksys-wrt1900ac"
#MACHINE = "mvebu-cortexa9"
#MACHINE = "marvell-armada"
#MACHINE = "marvell-armada-3720-db"
#MACHINE = "mvebu"
#MACHINE = "mvebu-cortexa9"

INHERIT += " openwrt-distro-defaults "
#IMAGE_FSTYPES += "iso"
```
[image]

kraj, 2024-02-20:
you might have to do more for building final images. We dont build openWRT images which maybe flashable directly using upgrade firmware, there must be more done for that. Flashing image built from qemuarm64 might be the first problem to solve. Use the correct machine and include right BSP layer which provides wrt1900ac machine definition.

MehrozMustafa, 2024-02-21:
> you might have to do more for building final images. We dont build openWRT images which maybe flashable directly using upgrade firmware, there must be more done for that. Flashing image built from [...]

Yeah, I understood the problem building qemuarm-based images is not the way to go.
https://docs.yoctoproject.org/transitioning-to-a-custom-environment.html#:~:text=Create%20your%20own%20layer%20for%20the%20BSP%20you%E2%80%99re%20going%20to%20use.
You are correct about including the correct BSP layer(s) but it's my first time doing something like this, so I'm unsure of how to add the correct BSD layer.
Are there some guide or tips I can follow to achieve my goal?