kraj/meta-openwrt #145: Building error (no __cxa_demangle in lstdc++), its a dependencies problem?

retsel89, 2024-03-13:
I get this message building for a raspberry pi zero wifi:

checking for __cxa_demangle in -lstdc++... no
| configure: error: __cxa_demangle not found in libstdc++, use --disable-demangler to disable demangler support.
| NOTE: The following config.log files may provide further information.
| NOTE: /home/lester/openwrt/openembedded-core/build/tmp-musl/work/x86_64-linux/elfutils-native/0.191/build/config.log
| ERROR: configure failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (virtual:native:/home/lester/openwrt/openembedded-core/meta/recipes-devtools/elfutils/elfutils_0.191.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1362 tasks of which 1230 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/home/lester/openwrt/openembedded-core/meta/recipes-devtools/elfutils/elfutils_0.191.bb:do_configure
Summary: There were 2 ERROR messages, returning a non-zero exit code.

kraj, 2024-03-13:
@retsel89 it seems to be an issue with build host provided libstdc++ I think. Since its a native recipe. I think adding `--disable-demangler` to elfutils native configure options is a solution. However it will be good to understand why you are seeing it. So please describe your setup a bit. Secondly, also post the config.log from elfuitls-native build tree, which will have more informaiton about the failure

retsel89, 2024-03-15:
I will close since i try on a fresh instalation and work perfect