kraj/meta-openwrt #148: Unable to switch to a newer version of openwrt branch through bbappend file.

attina, 2025-03-18:
Dear maintainer,

I want to switch to a newer version of openwrt branch 24.10, so I added base-files-scripts.bbappend file with the following content in my layer.
```
SRC_URI = "git://github.com/openwrt/openwrt.git;protocol=https;branch=openwrt-24.10;name=openwrt;destsuffix=git/openwrt"

OPENWRT_SRCREV = "5c533d7a898b6aed5af5c52f0aaf432d632281ab"
```
But It seems the SRC_URI didn't change, meanwhile the OPENWRT_SRCREV successfully change.

Here is the error prompt:
```
base-files-scripts-1.0-r0 do_fetch: Fetcher failure: Unable to find revision 5c533d7a898b6aed5af5c52f0aaf432d632281ab in branch openwrt-22.03 even from upstream
```

kraj, 2025-03-18:
you also need to change 

recipes-tweaks/base-files/base-files-scripts.bb

attina, 2025-03-18:
> you also need to change
> 
> recipes-tweaks/base-files/base-files-scripts.bb

Hi Kraj,

Change `recipes-tweaks/base-files/base-files-scripts.bb didn't` doesn't work. The following change can make bbappend overwrite SRC_URI working.
```
diff --git a/classes/openwrt-base-files.bbclass b/classes/openwrt-base-files.bbclass
index b680c37..4afda0a 100644
--- a/classes/openwrt-base-files.bbclass
+++ b/classes/openwrt-base-files.bbclass
@@ -15,7 +15,7 @@ OPENWRT_BASEPATH ?= "${S}/../git/openwrt"
 
 LIC_FILES_CHKSUM:append = " file://${OPENWRT_BASEPATH}/COPYING;md5=a8db84c7a073d2878849eee8eb0f5daa"
 
-SRC_URI:append = "\
+SRC_URI = "\
        git://github.com/openwrt/openwrt.git;branch=openwrt-22.03;protocol=https;name=openwrt;destsuffix=git/openwrt \
        "
 
(END)
```
but another error come up
```
CMake Error: The source directory "/build/tmp/work/cortexa72-cortexa53-crypto-poky-linux/uci/git/git" does not appear to contain CMakeLists.txt.
```

attina, 2025-03-18:
It seems the right way to do this is directly change
```
diff --git a/classes/openwrt-base-files.bbclass b/classes/openwrt-base-files.bbclass
index b680c37..9bd9625 100644
--- a/classes/openwrt-base-files.bbclass
+++ b/classes/openwrt-base-files.bbclass
@@ -7,7 +7,7 @@
 # Use this git SRCREV for all recipes that pull files out of openwrt repository
 # Equivalent to tag v22.03
 
-OPENWRT_SRCREV = "977f6f36a0121b213f9eee4a7deb7fdf073320e4"
+OPENWRT_SRCREV = "5c533d7a898b6aed5af5c52f0aaf432d632281ab"
 
 LICENSE:append = "& GPL-2.0-or-later"
 
@@ -16,7 +16,7 @@ OPENWRT_BASEPATH ?= "${S}/../git/openwrt"
 LIC_FILES_CHKSUM:append = " file://${OPENWRT_BASEPATH}/COPYING;md5=a8db84c7a073d2878849eee8eb0f5daa"
 
 SRC_URI:append = "\
-       git://github.com/openwrt/openwrt.git;branch=openwrt-22.03;protocol=https;name=openwrt;destsuffix=git/openwrt \
+       git://github.com/openwrt/openwrt.git;branch=openwrt-24.10;protocol=https;name=openwrt;destsuffix=git/openwrt \
        "
 
 SRCREV_FORMAT:prepend = "openwrt_"
diff --git a/recipes-tweaks/base-files/base-files-scripts.bb b/recipes-tweaks/base-files/base-files-scripts.bb
index 3022dfc..643531c 100644
--- a/recipes-tweaks/base-files/base-files-scripts.bb
+++ b/recipes-tweaks/base-files/base-files-scripts.bb
@@ -10,7 +10,7 @@ LIC_FILES_CHKSUM = "file://COPYING;md5=a8db84c7a073d2878849eee8eb0f5daa"
 
 FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}-scripts:"
 
-SRC_URI += "git://github.com/openwrt/openwrt.git;protocol=https;branch=openwrt-22.03 \
+SRC_URI += "git://github.com/openwrt/openwrt.git;protocol=https;branch=openwrt-24.10 \
            "
 
 SRCREV = "${OPENWRT_SRCREV}"
(END)
```

attina, 2025-03-18:
Hi Kraj,

Now, the build fail at
```
        # We want this to fail if Openwrt adds more to this dir, so no rm -rf
        rmdir ${STMP}/etc/uci-defaults
```
and the content of the folder `/etc/uci-defaults` is
```
11_network-migrate-bridges  50-root-passwd
```
Can you tell me how do we deal with these two files?